NorthernWidget-Skunkworks/Project-Okapi #6: Battery UVLO Issue

bschulz1701, 2020-01-04:
## Issue:
Battery (Li-Ion, VBat) can be drained beyond UVLO set by protection circuit due to the connection to the ADC through MUX. The battery (`LI+`) is connected directly to MUX (U38) instead of being run through protection circuit. This allows the parasitic drain from diode clamping in the MUX to drain the battery beyond the UV lockout.

## Potential Solution:
Simply connect the battery to the MUX through the protection system (`V_LI`) instead, which will protect it when it enters low voltage shutdown

bschulz1701, 2021-03-04:
Fixed as of v0.5

closed 2021-03-04 by bschulz1701

awickert, 2021-03-04:
@bschulz1701 hand-built and successfully tested 4d67928cf256c99434ec5b61966110642185817b.