NorthernWidget-Skunkworks/Project-Okapi #7: Output drain MOSFET issue

bschulz1701, 2020-01-04:
## Issue:
The MOSFET used on the production run of the `v0.4` boards (late 2019) used the 2N7002W-7-F transistor for `Q7`, this does not shut down the `5V_AUX` rail fast enough and can result in latching on and high current draw (~100mA) from the backup battery (`V_PRIME`) when in use. 

## Detailed Description:
[20200103185103557.pdf](https://github.com/NorthernWidget-Skunkworks/Project-Resnik/files/4021371/20200103185103557.pdf)

## Potential Solution: 
Replace MOSFET with 2N7002WT1G model instead. This device has been tested and results in fast enough shutdown to prevent any excess current draw from the backup battery. Not entirely clear (yet) why the part change fixes the issue, on paper they should be nearly equivalent. The updated part is ESD protected which could result in more reliable performance (inconsistent performance with previous part) and a slight change in diode forward voltage, more testing to be performed.

bschulz1701, 2021-03-04:
Note part replacement in Eagle file

awickert, 2026-11-06:
Checked against the Eagle files, 2026-11-06: not applied on the 2024 design. Q7 (drain on 5V_AUX, gate from U13) is still the 2N7002W-7-F that this issue says latches, and Q4 and Q8 carry the same number. The fix is a part number, 2N7002WT1G, and it blocks a first build. Moved from v0.6.0 to v0.7.0, with the BoM port (the part number goes in with the other field edits).

🤖 Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_017F2gr2axqf8HWh46zhsQyY

awickert, 2026-11-06:
What caused this, read from the v0.4 schematic (99b52bf) and the notes attached above (2026-11-06):

On v0.4, U38 was a TS5A21366 analog switch powered from 5V_AUX, with its switch inputs wired straight to the two batteries (LI+ and V_PRIME) to reach the ADC dividers. In sleep, POWER_EN is low, so the boost is off and Q7 is on, holding 5V_AUX at ground. The batteries then pushed current through the switch's input protection into its dead supply, which is 5V_AUX: the notes measure the rail at 1.3 V instead of 0, and the current went to ground through Q7. That is why it depended on V_PRIME being above V_bat and on the order the batteries changed in.

Why the 2N7002WT1G stopped it is not explained by the datasheets. It is the stronger switch (2.5 Ω maximum at a 4.5 V gate, against 7.5 Ω at 5 V for the 2N7002W-7-F), which, if anything, should draw more from such a leak; the other differences are a 2.5 V against 2.0 V maximum threshold and an ESD-protected gate. The swap most likely tipped a marginal condition one way.

The underlying cause is gone from the 2024 design: the TS5A21366 was removed, and the PAC1934, powered from 3V3_CORE, reads the batteries now. No part on today's 5V_AUX has a pin on a battery net. Q7 stays for its original job, a clean off-state: it takes 5V_AUX to 0 V within milliseconds when POWER_EN drops, so the ADS1115, the DAC, the level shifter and the Particle module's USB pin start from a full power-on reset at every wake, rather than from a rail left near 2.5 V by the reference (D6) and R48. With nothing leaking in, Q7 carries essentially no current in sleep. The part choice is therefore much less important than it was in 2020; the KiCad board carries the 2N7002WT1G as tested.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_017F2gr2axqf8HWh46zhsQyY

awickert, 2026-11-06:
Q4 and Q8 now carry the 2N7002WT1G too (Andy, 2026-11-06), so the three 2N7002s are one line on the BoM. Neither showed a fault; the change is for one part number and the part tested on this board. Each job has its gate at 3.3 V, and each load is small: Q4 pulls RST low against R52 (10 kΩ, 0.33 mA) when the Particle asks for a reset, and Q8 switches R70 (200 kΩ) on the charger's !SHDN divider from !CHRG. The onsemi part's higher maximum threshold (2.5 V against 2.0 V) still leaves far more drive than these loads need, and its gate leakage (150 nA maximum at 5 V) costs Q8 at most 0.15 V across its 1 MΩ pull-up, R55.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_017F2gr2axqf8HWh46zhsQyY

awickert, 2026-11-10:
Done in the KiCad design: Q7, with Q4 and Q8, carries the onsemi 2N7002WT1G (Project-Okapi 99ad985 and 18c6a63). The battery leak that made Q7 latch was removed in the 2024 design; the part stays for a clean 5V_AUX off-state.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-11-10 by awickert