NorthernWidget-Skunkworks/Project-Okapi #13: v1.0 Updates

bschulz1701, 2024-09-23:
### Improvement for `v1.0` design

- [ ] Switch to USB C
- [ ] Add LED system from Kestrel 
- [ ] Add CSA from Kestrel (PCA1934)
- [ ] Add FRAM 
- [ ] Add on-board sensors 
     - [ ] Light
     - [ ] Accel
     - [ ] Mag
- [ ] Investigate option for OTA support

awickert, 2026-11-06:
Checked 2026-11-06: nine of the ten items are on the 2024 design (USB-C, LED grid, PAC1934, FRAM, the three sensors, the SoM I2C buffer). The open item, solar charger performance, now has its own issues: #32 (the charger has no sense resistor) and #20 (closed, the MPPT point is right for the P106 panel). Left open for v1.0.0.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_017F2gr2axqf8HWh46zhsQyY

awickert, 2026-11-06:
Placed on a new v1.0.0 milestone (Andy, 2026-11-06). The one open item, solar charger performance, is tested on a v0.7.0 board by the MPPT and charger checks in `docs/v0.7-pre-build-checks.md`, with #32 and #30 behind it.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

https://claude.ai/code/session_017F2gr2axqf8HWh46zhsQyY